espressif/esp32-camera #800: ESP32 EV-VSYNC-OVF on esp_camera_init

Marthaarman, 2025-09-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I am trying to use a ov2640 with ESP32 (fire beetle). 
The camera that I got: https://domoticx.net/webshop/camera-module-ov2640-1600x1200-uxga/ - It has an internal clock so I'm not providing any clock signal to this camera. On an oscilloscope I found that this is true, the pixel clock is returned at around 24MHz by default.

Now I used this code:
```
#include <Wire.h>
#include <driver/i2s.h>
#include <Arduino.h>
#include <esp_camera.h>

camera_config_t config = {
  .pin_pwdn = -1,
  .pin_reset = -1,
  .pin_xclk = 4,
  .pin_sscb_sda = 21,    // Might vary
  .pin_sscb_scl = 22,    // Might vary

  .pin_d7 = 5,
  .pin_d6 = 13,
  .pin_d5 = 10,
  .pin_d4 = 9,
  .pin_d3 = 35,
  .pin_d2 = 34,
  .pin_d1 = 39,
  .pin_d0 = 36,

  .pin_vsync = 27,
  .pin_href = 26,
  .pin_pclk = 25,

  .xclk_freq_hz = 24000000,
  .ledc_timer = LEDC_TIMER_0,
  .ledc_channel = LEDC_CHANNEL_0,

  .pixel_format = PIXFORMAT_GRAYSCALE, // or RGB565/YUV422/...
  .frame_size = FRAMESIZE_QQVGA,
  .jpeg_quality = 12,
  .fb_count = 1,
  .fb_location = CAMERA_FB_IN_DRAM
};

void setup() {
  Serial.begin(2000000);
  delay(5000); // Let serial settle
  Serial.println("Setup done");
  // Initialize camera
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed: 0x%x\n", err);
    return;
  }
  Serial.printf("Camera init ready\n");
  sensor_t * s = esp_camera_sensor_get();
  // Example: divide clock by 4 (so output = XCLK / 4)
  s->set_reg(s, 0x11, 0xFF, 0x18);  // 0x11 = CLKRC, 0x03 => /4

  Serial.println("Camera ready.");
}

void loop() {
  camera_fb_t *fb = esp_camera_fb_get();
  if (!fb || fb->len != 320 * 240) {
    Serial.println("Frame error");
    if (fb) esp_camera_fb_return(fb);
    delay(2000);
    return;
  }

  // Send frame header: 'FRAM' + 4-byte length
  Serial.write("FRAM", 4);
  uint32_t len = fb->len;
  Serial.write((uint8_t*)&len, 4);

  // Send raw grayscale image bytes
  Serial.write(fb->buf, len);

  Serial.write("END", 3);

  esp_camera_fb_return(fb);
  // // delete fb;
  delay(1000);  // 1 second between frames
}

```

I noted that I cannot reupload new designs without disabling the camera first. Probably because of interrupt overload or pinout selection on my esp32? 

When I reconnect camera I get a continuous log stream with:
`cam_hal: EV-VSYNC-OVF`

Any clue what I might be missing here?

me-no-dev, 2025-09-03:
> the pixel clock is returned at around 24MHz by default.

This could be the issue. Do you have an option to lower the pixel clock?

> init doesn't seem to work at all.

What do you mean by that? Did you get `Camera init failed`?

Marthaarman, 2025-09-03:
> This could be the issue. Do you have an option to lower the pixel clock?

This is why I tried to write to register 0x11 which is the prescaler, lowering the pixel clock. 

[image]

> What do you mean by that? Did you get `Camera init failed`?
Well, I do not get the line `  if (err != ESP_OK) {
    Serial.printf("Camera init failed: 0x%x\n", err);
    return;
  }` printed, but I do not get this line printed either: `Serial.printf("Camera init ready\n");`

me-no-dev, 2025-09-03:
So first init with lower XCLK (the driver will try to start XCLK anyway). Lets see if you can get past init